QIDITECH/klipper #1: Klipper/Fluidd running on external system (X-Smart 3)

autoferrit, 2024-03-08:
As this not my first 3d printer,I already have klipper/fluidd running on a dedicated linux box that has plenty of power to manage multiple printers. Is it possible to run your klipper/fluidd a dedicated box instead of built in? or at least install your custom klipper/fluidd along an existing install to manage the printer? it would be nice to manage all my printers from one place.

I see that klipper can handle managing multiple printers but I don't know if that will work as easily since you have modified the source in your fork. Another reason for this is that my system is likely much more powerful than what is on the stock board. But maybe it doesn't matter?

Was hoping to get any advice or input on this.

CChen616, 2024-03-09:
It's a very interesting idea. Though we haven't done something like this before, however, I've gathered some related materials and discussions for you：

[https://www.klipper3d.org/FAQ.html#can-i-run-multiple-instances-of-klipper-on-the-same-host-machine](https://www.klipper3d.org/FAQ.html#can-i-run-multiple-instances-of-klipper-on-the-same-host-machine)
https://www.reddit.com/r/klippers/comments/ult5lg/good_guide_to_setting_up_a_second_instance_of/
https://www.reddit.com/r/klippers/comments/1agrl5c/running_multiple_printers_from_1_pc/

According to these materials, It seems possible to install multiple Klipper/Moonraker instances on your primary Linux system using Kiauh by selecting multiple instances. Then, move the original Klipper files to the new location created by Kiauh and overwrite them, modifying the corresponding configurations, such as MCU. However, be aware that the printer's built-in screen, which is controlled by the original Linux system's software, may become non-functional after migration.

 I'm sorry I can't be of more assistance at this moment. However if you make any progress, I'd be more than happy to follow up and learn more about it.

minus1ms, 2024-03-12:
Hi, im interested in this too, i dont care abot screen, i can use my old android phone as a klipperscreen, i just want the original klipper cuz your has some bugs, the problem is that in original klipper's configmenu i cant chose this custom bord that is here, can you pls tell me how do you build this klipper version? what do you select in the configmenu? also i want to know what is different about this klipper compared to the original @CChen616

minus1ms, 2024-03-12:
also im know how to program so maybe i will be able to modify latest original klipper to work for this board

CChen616, 2024-03-13:
> Hi, im interested in this too, i dont care about screen, i can use my old android phone as a klipperscreen, i just want the original klipper cuz your has some bugs, the problem is that in original klipper's configmenu i cant chose this custom bord that is here, can you pls tell me how do you build this klipper version? what do you select in the configmenu? also i want to know what is different about this klipper compared to the original @CChen616

Below is the firmware make config for the mainboard MCU:
[image]
And here is the firmware make config for the extruder's sub-MCU:
[image]

If you are not using a screen, then the printer should be able to operate normally with the original Klipper, but the `printer.cfg` may require some modifications. Also, be cautious of changes in the network environment that could prevent SSH connections to the mainboard. Additionally, alterations to the system that result in malfunctions are not covered under our warranty.

You can obtain the version of Klipper we are based on by using:
```bash
git clone https://github.com/Klipper3d/klipper.git

cd klipper

git checkout 0c74b3d8bfb8d06ac6ddc208a0a53bba235ee89d
```
Then, you can compare with the modified version.

minus1ms, 2024-03-13:
thank you, for the sub mcu i managed to do that by myself, but for main mcu i was confused because in printer cfg there is info that 48bit bootloader should be used, isnt it misleading?

minus1ms, 2024-03-13:
i have already updated my klippers everywhere except in main mcu but everything works like it shold, i have mainsail instead of fluidd and im happy with that

minus1ms, 2024-03-13:
can u pls provide me with the flashing instructions for the main MCU @CChen616

CChen616, 2024-03-14:
> can u pls provide me with the flashing instructions for the main MCU @CChen616

The instructions written in the configuration are somehow incorrect. After compiling the `klipper.bin` according to the above configuration, rename it to `X_4.bin` and move it to the SD card to be loaded into the machine at this location. 
[image]

Restart the machine and wait for MCU to be updated.

minus1ms, 2024-03-14:
Thank you very much!

autoferrit, 2024-04-02:
here is a community guide posted in the discord that walks through the process to install klipper 

https://github.com/Phil1988/FreeQIDI

i will be trying this hopefully this weekend