QIDITECH/klipper #5: TOGGLE_CHAMBER_FAN is poorly implemented

stew675, 2024-10-23:
See here: https://github.com/QIDITECH/klipper/blob/b90736975baeed82ff93d87c0260fd6b24889e7b/klippy/extras/chamber_fan.py#L31-L33

This creates the `TOGGLE_CHAMBER_FAN` custom g-code command, but there is no indication of whether the fan was on or off.

Within the `get_zoffet` macro definition (see here: https://github.com/QIDITECH/QIDI_PLUS4/blob/main/config/gcode_macro.cfg#L29-L45) we can see that `TOGGLE_CHAMBER_FAN` is called.  `get_zoffset` is called by the `G29` macro and others.

The problem is that this is subject to a functional race.  If the heater was active prior to `get_zoffset` being called, then the call to `M141 S0` will start to make the chamber fan shutdown.  Right now, the chamber fan will stay active for 60s after the chamber fan is turned off.

Now, if the chamber heater was already on during the nozzle wipe sequence, but had already reached temperature just prior to `get_zoffset` being called, then the chamber heater can still be hot enough to trip the chamber protection sensor in certain scenarios.  Some customers have seen this, as the chamber fan can be forced into the off set too soon.  It would be prudent to inject a `G4 P15000` command after the `M141 S0` command to ensure that the chamber heater unit has had sufficient time to cool down before the fan is turned off.

Additionally, the behavior of `TOGGLE_CHAMBER_FAN` is inherently relying on the internal state of `chamber_fan.py` to ensure that the fan, if initially off, won't turn on because the heater was disabled.  It's just bad programming practise as it leaves the fan in a non-determinant state.

I modified `chamber_fan.py` to give an explicit on/off gcode command like so:

```
        gcode.register_command("TOGGLE_CHAMBER_FAN", self.cmd_toggle_chamber_fan)
        gcode.register_command("ENABLE_CHAMBER_FAN", self.cmd_enable_chamber_fan)
        gcode.register_command("DISABLE_CHAMBER_FAN", self.cmd_disable_chamber_fan)
    def cmd_toggle_chamber_fan(self, gcmd):
        self.fan_on = not self.fan_on
    def cmd_enable_chamber_fan(self, gcmd):
        self.fan_on = True
    def cmd_disable_chamber_fan(self, gcmd):
        self.fan_on = False
```

It seems to me that `get_zoffset` wants to be calling `DISABLE_CHAMBER_FAN` at the start of the macro, and calling `ENABLE_CHAMBER_FAN` at the end of it, and that's the way it really should be, not potentially flipping the fan state in a non-deterministic fashion and trusting other code to handle it properly.

QIDITECH, 2024-10-28:
Thank you for your advice.

nicwilson58, 2024-11-14:
thanks Stew675, I agree and I have made these changes to mine