QIDITECH/klipper #7: adding filament change script

hosterweb1, 2024-10-26:
Hello. Please help me organize the process of changing the filament. As was done on my previous printer:
When it comes to changing the rod in gcode, the printer pauses and starts the filament change cycle. Unloads the old one, and gives a signal waiting for the new one to be loaded. After that, I manually remove the spool and put in a new one, load the filament into the extruder and press the button on the screen, and printing continues.
Plus 4 has commands m603 and m604, maybe I can implement it somehow with them?

QIDITECH, 2024-10-28:
You can edit both Gcodes in gcode_macro.cfg
[image]

hosterweb1, 2024-10-28:
> You can edit both Gcodes in gcode_macro.cfg
> _20241028084920.png (view on web)

Whhat i should edit there? Can you help?

QIDITECH, 2024-10-29:
You can directly edit M603 or M604 according to your needs
[image]

QIDITECH, 2024-10-30:
Sorry, this is not possible